limbo018/DREAMPlace #212: Errors while building

zk1056309462, 2025-03-04:
[ 63%] Building NVCC (Device) object dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o
cd /home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src && /usr/bin/cmake -E make_directory /home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/.
cd /home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src && /usr/bin/cmake -D verbose:BOOL=1 -D build_configuration:STRING=Release -D generated_file:STRING=/home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/./dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o -D generated_cubin_file:STRING=/home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/./dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o.cubin.txt -P /home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o.Release.cmake
-- Removing /home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/./dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o
/usr/bin/cmake -E rm -f /home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/./dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o
-- Generating dependency file: /home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o.NVCC-depend
/usr/bin/nvcc -M -D__CUDACC__ /home/zhuke/EDA/DREAMPlace/dreamplace/ops/dct/src/dst_cuda_kernel.cu -o /home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o.NVCC-depend -m64 -DTORCH_EXTENSION_NAME=dct_dst_kernel_cuda -DTORCH_VERSION_MAJOR=2 -DTORCH_VERSION_MINOR=0 -DENABLE_CUDA=1 -D_GLIBCXX_USE_CXX11_ABI=1 -Xcompiler ,\"-O3\",\"-DNDEBUG\" -gencode arch=compute_60,code=sm_60 -gencode arch=compute_61,code=sm_61 -gencode arch=compute_70,code=sm_70 -gencode arch=compute_75,code=sm_75 -gencode arch=compute_80,code=sm_80 --compiler-options -fPIC -DNVCC -I/usr/include -I/home/zhuke/EDA/DREAMPlace/thirdparty/flute -I/home/zhuke/EDA/DREAMPlace/thirdparty/OpenTimer -I/home/zhuke/EDA/DREAMPlace/dreamplace/ops/dct/.. -I/opt/miniconda3/envs/pytorch-2.0.0-zk/lib/python3.9/site-packages/torch/include -I/opt/miniconda3/envs/pytorch-2.0.0-zk/lib/python3.9/site-packages/torch/include/torch/csrc/api/include -I/home/zhuke/EDA/DREAMPlace/thirdparty/pybind11/include -I/opt/miniconda3/envs/pytorch-2.0.0-zk/include/python3.9
-- Generating temporary cmake readable file: /home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o.depend.tmp
/usr/bin/cmake -D input_file:FILEPATH=/home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o.NVCC-depend -D output_file:FILEPATH=/home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o.depend.tmp -D verbose=1 -P /usr/share/cmake-3.22/Modules/FindCUDA/make2cmake.cmake
-- Copy if different /home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o.depend.tmp to /home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o.depend
/usr/bin/cmake -E copy_if_different /home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o.depend.tmp /home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o.depend
-- Removing /home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o.depend.tmp and /home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o.NVCC-depend
/usr/bin/cmake -E rm -f /home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o.depend.tmp /home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o.NVCC-depend
-- Generating /home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/./dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o
/usr/bin/nvcc /home/zhuke/EDA/DREAMPlace/dreamplace/ops/dct/src/dst_cuda_kernel.cu -c -o /home/zhuke/EDA/DREAMPlace/build/dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/./dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o -m64 -DTORCH_EXTENSION_NAME=dct_dst_kernel_cuda -DTORCH_VERSION_MAJOR=2 -DTORCH_VERSION_MINOR=0 -DENABLE_CUDA=1 -D_GLIBCXX_USE_CXX11_ABI=1 -Xcompiler ,\"-O3\",\"-DNDEBUG\" -gencode arch=compute_60,code=sm_60 -gencode arch=compute_61,code=sm_61 -gencode arch=compute_70,code=sm_70 -gencode arch=compute_75,code=sm_75 -gencode arch=compute_80,code=sm_80 --compiler-options -fPIC -DNVCC -I/usr/include -I/home/zhuke/EDA/DREAMPlace/thirdparty/flute -I/home/zhuke/EDA/DREAMPlace/thirdparty/OpenTimer -I/home/zhuke/EDA/DREAMPlace/dreamplace/ops/dct/.. -I/opt/miniconda3/envs/pytorch-2.0.0-zk/lib/python3.9/site-packages/torch/include -I/opt/miniconda3/envs/pytorch-2.0.0-zk/lib/python3.9/site-packages/torch/include/torch/csrc/api/include -I/home/zhuke/EDA/DREAMPlace/thirdparty/pybind11/include -I/opt/miniconda3/envs/pytorch-2.0.0-zk/include/python3.9
/usr/include/c++/11/type_traits(1406): error: type name is not allowed

/usr/include/c++/11/type_traits(1406): error: type name is not allowed

/usr/include/c++/11/type_traits(1406): error: identifier "__is_same" is undefined

/usr/include/c++/11/type_traits(3251): error: type name is not allowed

/usr/include/c++/11/type_traits(3251): error: type name is not allowed

/usr/include/c++/11/bits/stl_pair.h(460): error: argument list for class template "std::pair" is missing

/usr/include/c++/11/bits/stl_pair.h(460): error: expected a ")"

/usr/include/c++/11/bits/stl_pair.h(460): error: template parameter "_T1" may not be redeclared in this scope

/usr/include/c++/11/bits/stl_pair.h(460): error: expected a ";"

/usr/include/c++/11/bits/stl_vector.h(1876): error: argument list for class template "std::vector" is missing

/usr/include/c++/11/bits/stl_vector.h(1876): error: expected a ")"

/usr/include/c++/11/bits/stl_vector.h(1876): error: template parameter "_InputIterator" may not be redeclared in this scope

/usr/include/c++/11/bits/stl_vector.h(1877): error: expected a ";"

/usr/include/c++/11/utility(312): error: pack expansion does not make use of any argument packs

/usr/include/c++/11/utility(334): error: pack expansion does not make use of any argument packs

/usr/include/c++/11/utility(334): error: expected a ">"
          detected during instantiation of type "std::make_integer_sequence<std::size_t, _Num>" 
(343): here

/usr/include/c++/11/array(267): error: argument list for class template "std::array" is missing

/usr/include/c++/11/array(267): error: expected a ")"

/usr/include/c++/11/array(267): error: template parameter "_Tp" may not be redeclared in this scope

/usr/include/c++/11/array(268): error: expected a ";"

/usr/include/c++/11/tuple(948): error: argument list for class template "std::tuple" is missing

/usr/include/c++/11/tuple(948): error: expected a ")"

/usr/include/c++/11/tuple(948): error: template parameter "_UTypes" may not be redeclared in this scope

/usr/include/c++/11/tuple(948): error: expected a ";"

/usr/include/c++/11/tuple(950): error: argument list for class template "std::tuple" is missing

/usr/include/c++/11/tuple(950): error: "pair" is not a function or static data member

/usr/include/c++/11/tuple(952): error: argument list for class template "std::tuple" is missing
[... 143 lines cut ...]

/usr/include/c++/11/bits/unordered_map.h(1159): error: expected a ")"

/usr/include/c++/11/bits/unordered_map.h(1159): error: "initializer_list" is not a function or static data member

[... cut ...]
make[2]: *** [dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/build.make:87: dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/src/dct_dst_kernel_cuda_generated_dst_cuda_kernel.cu.o] Error 1
make[2]: Leaving directory '/home/asd/EDA/DREAMPlace/build'
make[1]: *** [CMakeFiles/Makefile2:4025: dreamplace/ops/dct/CMakeFiles/dct_dst_kernel_cuda.dir/all] Error 2
make[1]: Leaving directory '/home/asd/EDA/DREAMPlace/build'
make: *** [Makefile:139: all] Error 2

Hi, when I compile DREAMPlace I get the above errors, please tell me how to fix these errors, thanks! BTW, my OS is Ubuntu 22.04.3 LTS and gcc version is 7.5, CUDA version is 11.8.

limbo018, 2025-03-04:
Did you set ```CMAKE_CXX_ABI``` to 1? I think pytorch is compiled with ```_GLIBCXX_USE_CXX11_ABI=0```.

zk1056309462, 2025-03-04:
Thank you for your patient reply. I set the reference of _GLIBCXX_USE_CXX11_ABI to 1, but this error also occurred when its value was 0.

zk1056309462, 2025-03-04:
Yes, I will try to change the CUDA vision. Thank you very much for your reply.

Iampooo, 2025-05-27:
I encounter the same error. I'm using CUDA 11.5 and gcc version 7.5 and set _GLIBCXX_USE_CXX11_ABI=0