limbo018/DREAMPlace #214: gift init error

shypark98, 2025-04-15:
Dear Prof. Lin,

I tried enabling GiFt initialization in DREAMPlace by setting gift_init_flag to 1 in DREAMPlace/install/dreamplace/params.json. However, when I run the placer, it results in a segmentation fault. When gift_init_flag is set to 0, the flow runs correctly without any issues. 
I also tried modifying the gift_init_scale value, but the segmentation fault still occurs.

What could be the possible cause of this segmentation fault, and how can I go about debugging it?

[INFO   ] DREAMPlace - reading benchmark takes 0.103866 seconds
[INFO   ] DREAMPlace - reading database takes 0.10 seconds
[INFO   ] DREAMPlace - move cells to the center of layout with random noise
[INFO   ] DREAMPlace - non-linear placement initialization takes 0.42 seconds
[INFO   ] DREAMPlace - Global placement: 512x512 bins by default
[INFO   ] DREAMPlace - 512x512 bins #bins 512x512, bin sizes 0.238281x0.234375, max_num_bins = 23, padding = 0
[INFO   ] dreamplace.ops.electric_potential.electric_potential - regard 0 cells as movable macros in global placement
[WARNING] DREAMPlace - quadratic density penalty is disabled, density weight update is forced to be based on HPWL
[INFO   ] dreamplace.ops.gift_init.gift_init - Construct adjacency matrix using clique model
Segmentation fault (core dumped)

limbo018, 2025-04-15:
Which case did you run?

shypark98, 2025-04-15:
oh sorry, it was ispd19_test1

limbo018, 2025-04-15:
Please pull the commit (https://github.com/limbo018/DREAMPlace/commit/4e9452518fe06693e418cbff9137ba050d6d7bb8). The bug should be fixed.

shypark98, 2025-04-15:
I just checked it works well. Thank you.

closed 2025-04-15 by shypark98