limbo018/DREAMPlace #217: I meet error : "malloc(): corrupted top size   Aborted (core dumped)" while running Placer.py on iccad2015.ot/superblue1.json

lfy2947590638, 2025-04-23:
[image]
I meet error when I run **python dreamplace/Placer.py test/iccad2015.ot/superblue1.json**

limbo018, 2025-04-24:
Do all superblue cases fail or just superblue1?

lfy2947590638, 2025-04-24:
> Do all superblue cases fail or just superblue1?

Thanks for answering! I find that all superblue cases fail, and it seems that they fail at the same processing node?

[image]

enzoleo, 2025-05-04:
Would you check whether you are using the modified ICCAD2015 [here](https://github.com/limbo018/DREAMPlace/blob/master/benchmarks/iccad2015.ot.md)? The original benchmark does not work well with OpenTimer.

lfy2947590638, 2025-05-11:
> Would you check whether you are using the modified ICCAD2015 [here](https://github.com/limbo018/DREAMPlace/blob/master/benchmarks/iccad2015.ot.md)? The original benchmark does not work well with OpenTimer.

Yes, I downloaded ICCAD2015.ot benchmarks from this link.

lfy2947590638, 2026-01-21:
> [@lfy2947590638](https://github.com/lfy2947590638) Bro, have you solved this issue? I found that the error is actually coming from flute in my environment.
> 
> <img alt="Image" width="1192" height="293" src="https://private-user-images.githubusercontent.com/107295305/538577657-75c70622-2fed-403a-bfd8-9eefdb8e58fe.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjkwMTY0NTksIm5iZiI6MTc2OTAxNjE1OSwicGF0aCI6Ii8xMDcyOTUzMDUvNTM4NTc3NjU3LTc1YzcwNjIyLTJmZWQtNDAzYS1iZmQ4LTllZWZkYjhlNThmZS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjYwMTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI2MDEyMVQxNzIyMzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lYmE5YTIwMjE0OTA0ZWU1MDkxZjY4NzcxYzBhZjhjOTFiYjYyYjNlYjQ1MzJkNmI0ZDc0ZDc5OTZmMWI3YzExJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.EzbKJMHMosOhxpyW-krqc5zDs-GL4tzI2Ww0IRWZvQ8">

Thank you for reaching out and sharing your finding about the flute library. Honestly, I never managed to resolve this issue myself, and I hadn’t identified flute as the source of the problem before. Since I’m no longer actively working in this research direction, I’m afraid I can’t offer further help here.

I’d suggest:

Checking if there are any updates or patches related to flute in the DREAMPlace repository.

Opening a new issue specifically detailing the flute-related crash with your logs—this might attract attention from maintainers or other users who’ve solved it.

Looking into whether rebuilding flute or adjusting memory/thread settings resolves the corruption.

Wishing you the best in debugging this.